peninquen/Modbus-Energy-Monitor-Arduino #10: Me puedes explicar  (*mbm).connect(this);

cosmopaco, 2015-12-16:
Me puedes explicar como funciona o donde puedo leerlo.
No lo habia visto nunca!!!!!

peninquen, 2015-12-16:
Pues es cosecha propia (y funciona).
He creado un array en ModbusMaster con punteros a los objetos que va a muestrear para procesarlos consecutivamente.
En el constructor del objeto ModbusSensor paso su puntero (this) como parametro.
Si ves la función connect() lo pongo en la pila y mantengo totalSensor como valor límite.
Seguro que hay otras estructuras, pero esta me ha parecido la más C++ ¿alguna sugerencia?

cosmopaco, 2015-12-16:
No llego a ese nivel ;). 
Soy autodidacta en programación, y me ha sorprendido. Los punteros todavía los entiendo pero puntero a objeto se me escapa.

Algo nuevo para aprender.!!!

peninquen, 2015-12-16:
Yo también soy autodidacta, pero parece que mi libro de 'estructuras y algoritimos' caló profundo ;). Ya de herencias voy más flojo.
Yo he aprendido mucho viendo las librerías originales de arduino, muy recomendable.

cosmopaco, 2015-12-18:
Cierro.
Lo estudiare con calma.

closed 2015-12-18 by cosmopaco