peninquen/Modbus-Energy-Monitor-Arduino #11: desbloquear arduino durante comunicaciones - event-driving programming

peninquen, 2015-12-17:
El programa de arduino debe estar orientado a la gestión de eventos, si el envío-recepción de mensajes dura 40- 70 ms durante ese tiempo no puede ejecutar ningún otro módulo (salvo por interrupciones).
objetivo: modificar las funciones sendFrame() y readBuffer() para enviar/leer un byte, retornar y mantener el _state hasta que haya terminado de enviar/recibir el frame.
modo:
- sendFrame() función que devuelve un boolean con el estado de proceso, solo cuando sea true cambiar de estado. No vuelve a enviar un byte hasta pasado T1_5
- crear un nuevo _state llamado IDLE para esperar TIMEOUT desde que termina de enviar la trama hasta que recibe el primer byte de respuesta.
- readBuffer() funcion que devuelve un boolean, 'false' cuando no ha pasado T3_5 desde el ultimo byte recibido. Cuando pasa T3_5 procesa el buffer y devuelve true

cosmopaco, 2015-12-17:
Creo que estas equivocado.
Como has medido los tiempos???

En el envió/recepción se encarga la libreria Arduino Serial. En el caso de envió se carga sobre el buffer del Serial y la función se encarga del envió. En la recepción ocurre lo mismo es la librería serial la que se encarga de recibir y guardar en su buffer hasta que sean leído. Por la libreria Modbus.

peninquen, 2015-12-18:
Si miras dentro de la función sendFrame() y receiveBuffer() se manda al monitor serial el millis() de comienzo y el millis() de final:
200 MASTER: 01 04 00 00 00 02 71 CB    248
333  SLAVE: 01 04 04 43 59 00 00 3E 13 397
mientras está dentro de la función no puede hacer otra cosa, así que está retenido.
Probaré a si sacando la función .flush() se acorta el tiempo

peninquen, 2015-12-18:
Tengo una curiosidad, el protocolo Modbus habla de los tiempos entre caracteres, ¿cómo lo hace?
- es standard,
- el SDM lo acepta aunque no lo respete
??

cosmopaco, 2015-12-18:
El tiempo entre caracteres no se cumple. Por ese motivo a estas librerias se les llama modbus sobre serial.
Se tendria que trabajar a bajo nivel para cumplirlo.

cosmopaco, 2015-12-18:
El tema de los tiempos yo no lo veo así.

Se muestra cuando se ha escrito en el bufer de serial, y el momento en que la libreria serial devuelve la lectura. 

El unico "bloqueo" puede producirlo Serial.flush ().   Esta función  espera ha que se vacie el buffer de salida (depende de la librería Stream) y que haya salido el último bit.(dependencia libreria HardwareSerial)

peninquen, 2015-12-18:
El problema es que debemos mantener el TxEnablePin en HIGH para asegurar que el MAX485 está en modo Tx, eso obliga a utilizar flush(). Otra posibilidad es que fuera de la función sendFrame() chequee cuándo está vacio el buffer y entonces desactiva el pin.

No entiendo tu comentario anterior, los tiempos son dentro de función sendFrame y readBuffer, por tanto no puede ejecutar ningún otro evento tipo 'leer pulsador'

peninquen, 2015-12-18:
No encuentro ninguna función en la librería Stream ni Serial que devuelva el numero de caracteres pendientes de enviar, todas las funciones hacen referencia a los pendientes de leer. Voy a mirar las tripas de Serial y de Stream.
Otra posibilidad es activar un temporizador (función del Baud rate y número de caracteres enviados) que mantenga el pin HIGH. De hecho, el pin en HIGH debe estar asociado al estado SENDING

cosmopaco, 2015-12-18:
availableForWrite(void)  ??? 

```
   if ((*_MBSerial).availableForWrite()== SERIAL_TX_BUFFER_SIZE){        
        Serial.flush();
        digitalWrite(_TxEnablePin, LOW);    
      }
      else
        return false;
```

peninquen, 2015-12-18:
Estamos con el mismo problema, la función flush bloquea hasta que no se vacía, y no he visto ninguna manera de crear availableForWrite() 

Puede que a bajo nivel haya un bit del UART que indique buffer vacío

cosmopaco, 2015-12-18:
availableForWrite() ya existe !!!

cosmopaco, 2015-12-18:
Esto repercute #12 :flushed:

peninquen, 2015-12-18:
Creo que devuelve el número de bytes disponibles para escribir:
``` c++
case RECEIVING:
   if ((*_MBSerial).availableForWrite() == 0) {        
//        Serial.flush();
        digitalWrite(_TxEnablePin, LOW);    
    }
    else
        return false;
```
podemos prescindir del flush() 
Lo que sí haría es crear otro estado distinto 
SENDING (controla indexSensor y sendFrame())
IDLE (espera a vaciar el buffer de escritura)
RECEIVING (espera a recibir la respuesta y la procesa o TimeOut)
WAITING_TO_SEND (espera a que se recupere el Bus T3.5, o el SDM)
WAITING_NEXT_POLL (periodo entre muestreos de lotes)

la estructura del 'switch(_state)' queda más clara

cosmopaco, 2015-12-18:
Creo que si devuelve el numero de bytes disponibles para escribi, quiere decir que cuando ha acabado de enviar la trama, tiene que devolver el tamaño total del buffer disponible.
Aunque no entiendo eso de 
if (head >= tail) return SERIAL_TX_BUFFER_SIZE - 1 - head + tail;
  return tail - head - 1;

peninquen, 2015-12-18:
Es un buffer circular, cuando la cabeza supera el SERIAL_TX_BUFFER_SIZE el tamaño pendiente no es 'head - tail -1' (supongo que el -1 es porque apunta al siguiente disponible

peninquen, 2015-12-18:
Tienes razón, devuelve la memoria disponible para escribir

cosmopaco, 2015-12-18:
Al final me comprare un cacharro nada mas que para aprender. JAJAJAJA
Estoy mirando las librerias ESP8266 y creo que son "equivalentes", Seria la impresionante hacerlo compatible con ESP.

peninquen, 2015-12-18:
Sé que lo tienes en mente, imagina una caja en carril DIN con su mochila ESP8266 transmitiendo los datos !! 
Lo más fácil sería hacer 
if (_tx_buffer_head == _tx_buffer_tail) 
¿HERENCIA?
¿friend class?

peninquen, 2015-12-18:
¿dónde estás mirando las librerias del ESP826?

cosmopaco, 2015-12-18:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/HardwareSerial.cpp

Lo de herencias ni idea. Aunque todo se aprende...

peninquen, 2015-12-19:
(http://c.conclase.net/curso/?cap=032#inicio)
Hacer la clase friend no funciona. Requiere modificar la librería HardwareSerial.
En cambio podemos hacer ModbusMaster heredera de HardwareSerial, con lo que pasa a tener acceso a todas las variables 'protected' _tx_buffer_head _tx_buffer_tail.
No sería necesario un buffer modbus teniendo el buffer Serial
Creo que es un buen ejemplo para aprender la herencia. Y la mejor manera de aprender es con un ejemplo práctico que motive...
¿cómo lo ves?

cosmopaco, 2015-12-20:
Me parece  bien. Puede haber un inconveniente,  seria portable a ESP8266?

Lo idoneo seria poder modificar de alguna manera la funcion _tx_udr_empty_irq . Superponer?
if (_tx_buffer_head == _tx_buffer_tail) {
    // Buffer empty, so disable interrupts
    cbi(*_ucsrb, UDRIE0);
    CAMBIAR ESTADO RECIBIR.
    PIN RE/DE LOW
  }

Aseguramos totalmente que el cambio del pin se hace a tiempo, sin la incertidumbre de qué el loop principal tenga retrasos en llamar a modbus.avalaible.

De esta manera aseguramos que

cosmopaco, 2015-12-20:
Que te parece el tema de la fundacion?  http://www.helianto.eu/

peninquen, 2015-12-20:
He visto la página y no es nada clara, no obstante nuestro trabajo aquí es desinteresado y open software. 
Sí me parece interesante como pulso de por dónde van los intereses de la gente, y esos son los sistemas WiFi y nodeMCU en particular. Todavía no lo he tratado pero le veo más posibilidades que a nRF.
Como te decía en el mensaje de arriba, desarrollar una mochila al SDM y poder ver en una app tu consumo actual y el histórico me parece algo muy atractivo.

peninquen, 2015-12-21:
Creo que vamos a tener problemas con la función
``` if (_tx_buffer_head == _tx_buffer_tail) {
// Buffer empty, so disable interrupts
cbi(*_ucsrb, UDRIE0);
CAMBIAR ESTADO RECIBIR.
PIN RE/DE LOW
``` 
Con esta comprobación sabemos cuando se ha liberado el buffer de TX, pero el último byte está en el registro UART y todavía no ha salido. La función Flush hace la misma comprobación, no sirve, así que solo queda fijar un timer desde la salida del dato del buffer hasta la desactivación del pin .Esto es lo que el protocolo llama T3.5, asegurar un vacío desde la salida de la trama hasta que el esclavo contesta. 
Propongo calcular T3_5 al definir el baudrate y marcar un temporizador en microsegundos. En caso de que el tiempo sea pequeño podemos utilizar un delayMicroseconds()

peninquen, 2015-12-21:
Solución al tratamiento durante el estado recibiendo trama del esclavo, marcar un temporizador cada vez que se recibe un caracter, si se supera el inter-caracter T1.5 (medido como T2.5 incluyendo el tiempo de lectura de un caracter) se considera que se ha terminado el frame y se procesa. permite leer todos los tamaños.
```
    case RECEIVING:

      if (!(*_MBSerial).available() && (millis() - sendMillis > _timeOut)) {
        (*_mbSensorPtr).putStatus(MB_TIMEOUT);
        indexSensor++;
        _state = SEND;
        return false;
      }

      if ((*_MBSerial).available() > frameSize) {
        frameSize++;
        tMicros = micros();
      }
      else {
        if (micros() - tMicros > _T2_5) {
          readBuffer(frameSize);
          while ((*_MBSerial).available()) (*_MBSerial).read(); // clean RX buffer
          MODBUS_SERIAL_PRINTLN((*_mbSensorPtr).getStatus(), HEX);
          indexSensor++;
          receiveMillis = millis(); //starts waiting interval to next request
          _state = IDLE;
        }
      }
      return false;
```

peninquen, 2015-12-21:
Esta es menos elegante, si queremos no utilizar delay() habría que definir otro estado
```
    case SENDING:

      if ((*_MBSerial).availableForWrite() == SERIAL_TX_BUFFER_SIZE) { //TX buffer empty
        delayMicroseconds(_T2_5); // time to be sure last byte sended
        digitalWrite(_TxEnablePin, LOW);
        sendMillis = millis(); //starts  slave's timeOut
        _state = RECEIVING;
        frameSize = 0;
      }
      return false;
```

peninquen, 2015-12-22:
Cierro este issue, creo que se ha alcanzado un nivel de no-blocking aceptable <2ms
[Monitor Serial debug.txt](https://github.com/peninquen/Modbus-Energy-Monitor-Arduino/files/69839/Monitor.Serial.debug.txt)

closed 2015-12-22 by peninquen