peninquen/Modbus-Energy-Monitor-Arduino #12: El nombre del estado standby no es descriptivo.

cosmopaco, 2015-12-17:
Creo que se debería renombrar como WAITING_FOR_TURNAROUND

peninquen, 2015-12-18:
¿Algo como 'WAITING_NEXT_POLL'?

cosmopaco, 2015-12-18:
Adjudicado.!!!

Esto plantea volver ha definir lo mejor posible la máquina de estado.
#define IDLE
#define SEND
#define SENDING   
#define RECEIVING 
#define WAITING_NEXT_POLL  

Lo he escrito a lo loco,

peninquen, 2015-12-18:
No lo había leido y escrito algo similar en el otro hilo!!!

peninquen, 2015-12-18:
¿Lo dejamos así?

SEND (controla indexSensor y sendFrame())
SENDING  (espera a vaciar el buffer de escritura)
RECEIVING (espera a recibir la respuesta y la procesa o TimeOut)
IDLE (espera a que se recupere el Bus T3.5, o el SDM)
WAITING_NEXT_POLL (periodo entre muestreos de lotes)

closed 2015-12-18 by peninquen