peninquen/Modbus-Energy-Monitor-Arduino #16: Eliminar la entrada de puerto serial del constructor modbusSensor.

cosmopaco, 2015-12-20:
Ya hay una entrada en modbusSerial es confuso!!

Si la intención  es que puedan usarse mas de un bus sobrecargar la función.

peninquen, 2015-12-20:
¿utilizar directivas del precompilador para saber en el constructor del modbusSensor dónde esta el modbusMaster al que colgarse?

peninquen, 2015-12-31:
En la librería .cpp he cambiado mbSerial por hwSerial para evitar confusiones con el objeto MBSerial de la clase modbusMaster.
voy a probar a definir en librería el objeto ***MBSerial***, sobrecargar las funciones de sensores para que por defecto se conecten y desconecten del array

peninquen, 2016-01-05:
Dejo un solo objeto MBSerial, no parece lógico gestionar más de un canal modbus, y elimino las referencias a la instancia de modbusMaster

closed 2016-01-05 by peninquen