peninquen/Modbus-Energy-Monitor-Arduino #19: formato para Base de datos

peninquen, 2015-12-20:
Aunque se sale del objeto de la librería, el siguiente paso es definir el tipo de base de datos a la que enviar los datos, definir la trama a enviar:
Fecha/hora: ¿AAAMMDDhh:mm:ss?
valor: ¿float? ¿entero con parte decimal? ¿cadena ASCII?
etiqueta del valor:¿mandamos una ristra CSV y el orden define qué parametro es? ¿cada parametro va acompañado de su etiqueta?
No tengo experiencia en web server ni en SQL, será lo siguiente...
Mi idea es introducirme en el funcionamiento de la raspi y ver diferentes opciones, una que he encontrado es cacti
http://trasteandoarduino.com/2014/04/27/monitorizacion-del-sistema-con-cacti-y-arduino-parte-1-instalacion/

¿Alguna experiencia o idea?

cosmopaco, 2015-12-20:
Que tal http://emoncms.org/

peninquen, 2015-12-22:
Parece una opción interesante, en principio orientada a su producto, pero abierta a aceptar otras entradas de datos. Sobre todo el uso que le da en el foro italiano muy interesante.

helianto, 2015-12-29:
Yo lo voy a usar con ese planteamiento y programa (he estado revisando todas las opciones posibles y es la mas flexible y abierta de todas), ademas de posiblemente enviar una "copia" a thingspeak con algunos datos (por si las moscas).

helianto, 2015-12-29:
Otra opcion interesante es usar mensajes mqqt, pero eso añadiria complicacion al sistema (si bien con mqqt y node-red se pueden hacer muchas cosas)

helianto, 2016-01-05:
Curioso el sistema de esp-easy, meten todos los protocolos...y luego tu eliges:

http://www.esp8266.nu/index.php/EasyProtocols