peninquen/Modbus-Energy-Monitor-Arduino #24: Incorporar manuales

peninquen, 2015-12-22:
Cosmopaco, Puedes volver a intentar fusionar los manuales? a ver si aprendo GitHub...

cosmopaco, 2015-12-22:
Llevo 2 horas intentando entender gjihub.
Me crea conflictos que no entiendo y no puedo enviar nada..

cosmopaco, 2015-12-23:
Logrado !!!!
Felices Fiestas.

helianto, 2016-01-05:
[SDM530M Three Phase Multi Function Din Rail Energy Meter user manual.pdf](https://github.com/peninquen/Modbus-Energy-Monitor-Arduino/files/78649/SDM530M.Three.Phase.Multi.Function.Din.Rail.Energy.Meter.user.manual.pdf)

Manual del 530... es raro que este modelo (mucho mas barato que el 630) tenga mas opciones de lectura que el 630 pues incluye datos acumulados por meses y muchos mas

helianto, 2016-01-05:
Por cierto, he solicitado a la empresa que me envien manuales actualizados y protocolos actualizados de sus productos... veremos si contestan.

peninquen, 2016-01-05:
Hecha un vistazo al catálogo 2015, No merece la pena preocuparse por modelos antiguos.

Ahora mismo el programa puede configurar y leer cualquier modelo, el siguiente objetivo es probar el resto de marcas

helianto, 2016-01-05:
En cuanto lo tenga te digo (el caso es que en la web venden todos).

P.D: puedes darme tu direccion de correo?, enviamela por favor a energia@helianto.eu

cosmopaco, 2016-01-05:
Ando con falta de tiempo. Aunque  sigo el gran trabajo de Peninquen, no tan a fonfo como me gustaria.

Creo que la libreria ha madurado lo suficiente como para intentar hacer pruebas con nodemcu.

Otro tema es documentar como se utiliza.

helianto, 2016-01-05:
Te comento lo mismo que a peninquen, por favor contactar conmigo por e-mail: energia@helianto.eu

peninquen, 2016-01-06:
Ahora que he terminado la programación gruesa paso a depurar, poner comentarios, ajustar nombres de variables para que sean más descriptivos, y documentar el uso.

peninquen, 2016-01-10:
He preparado un manual de usuario de la librería en la página principal. Falta cargar diagramas, y detallar los argumentos de las funciones.

helianto, 2016-01-10:
Menudo curro... felicidades...
Varias cosas: en breve (espero) me llega el sdm630 (bueno y alguno mas, pero esos ya son de pulsos...), en cuanto lo tenga puedo hacer pruebas con el 120 y el 630 con esp8266.

2º. Confirmar que el esp8266 serial port no es compatible con 5v, por lo que es necesario un level shifter si usamos el max485 (que requiere 5v de alimentacion) (tengo varios modulos por aqui asi que podra hacer pruebas).

3º. Si se quiere usar con un rtc, yo recomiendo esta libreria http://giltesa.com/2012/09/02/libreria-gds1307-para-rtc (que ademas acaba de incorporar unas funciones que le consulté recientemente), es con diferencia la mejor libreria de rtc que he visto, pues incorpora elementos como el cambio de hora verano/invierno (muy muy practico), comprobar si una fecha es posterior, anterior o entre unos valores, etc.

cosmopaco, 2016-01-10:
Enhorabuena gran trabajo  Haber si esta tarde saco un rato para mirarlo !!!!

peninquen, 2016-01-11:
alguna librería RTC habrá que utilizar, pero creo que en sentido inverso al que plantea @giltesa, el registro de datos se tiene que hacer con un registro de tiempos único' eso nos lleva a 'timestamps' únicos
https://tools.ietf.org/html/rfc3339
con un formato 'YYYY-MM-DDTHH:MM:SS para asegurar coherencia en la BD, después en la presentación se pueden realizar correcciones DST.

helianto, 2016-01-13:
En general es correcto lo que indicas, pero para el caso que quiero usar yo no... te explico: quiero hacer que desde arduino se mande una señal a iftt, cual es el problema, que ifft solo acepta una condicion (no acepta condiciones encadenadas), asi que una parte de la logica la tengo que realizar en el propio arduino (entre la que esta la logica "temporal"; es decir, para entenderlo: si la hora es entre las 9 de la noche y las 7 de la mañana, y el consumo electrico medio de las ultimas 2 horas ha sido superior a xxx Watt, envia un mensaje a iftt..., con estas premisas la libreria de giltesa me ahorra muchismo trabajo (pues incorpora todos los elementos que necesito)... ahora bien, reconozco que me sirve a mi... para el uso habitual no es necesario (aunque de hecho, para el uso habitual salvo que usemos un log en local no es necesario siquiera rtc, pues la propia bbdd nos metera el timestamp del sistema.

peninquen, 2016-01-13:
Yo sigo dándole vueltas al volcado de datos asíncrono, en ese caso necesito que los datos lleven su timestamp.

Para tu caso, ¿vas a incorporar un RTC al ESP?, ¿No te sería más práctico pedir la hora a red?  así  de pronto he encontrado 
http://polaridad.es/sincronizacion-horaria-wifi-esp8266/
http://www.prometec.net/ntp/

Te ahorras un módulo, batería y pines a los que conectar. Si miras dentro de la librería verás que la parte interesante son las funciones ``zeller()`` que se definen en el apéndice B
https://tools.ietf.org/html/rfc3339#appendix-B. y ``checkDST()``

helianto, 2016-01-13:
(yo juraria que habia respondido a esto, pero no encuentro el mensaje por ningun lado)...

En mi caso no tengo claro que voy a hacer pues para recogida de datos no me hace falta el rtc (la base de datos lo hace por mi), pero si lo necesito para el sistema de alertas (otra opcion es hacerlo desde el servidor, pero me gusta la sencillez de los microcontroladores). Mi problema con las librerias de ntp es el cambio horario invierno/verano (si te fijas, ninguno de los 2 que me has indicado contempla el cambio, y para sistemas a largo plazo (como los que tengo en mente) no son viables, de todos modos revisaré el tema mas a fondo.

peninquen, 2016-01-13:
Rutina de ajuste DST, sincronizar el reloj todos los días a las 02:00 con NTP http://forum.arduino.cc/index.php?topic=40286.0

helianto, 2016-01-13:
Me gusta tb esta opcion: https://gist.github.com/bbx10/149bba466b1e2cd887bf

helianto, 2016-01-18:
Me han enviado 2 manuales... uno ya estaba actualizado, el otro es la version 1.3 del modbus para el SDM630... lo pongo aqui (he pedido que me envien tb el del 530)

[EASTRON SDM630 Modbus protocol  V1.3.pdf](https://github.com/peninquen/Modbus-Energy-Monitor-Arduino/files/94176/EASTRON.SDM630.Modbus.protocol.V1.3.pdf)