peninquen/Modbus-Energy-Monitor-Arduino #31: Las clases sobrecargadas enlazadas ahorra tamaño Sketch.

cosmopaco, 2016-01-05:
Tamaño original compilación actual.
Sketch uses 8.956 bytes (3%) of program storage space. Maximum is 253.952 bytes.
Global variables use 767 bytes (9%) of dynamic memory, leaving 7.425 bytes for local variables. 
Maximum is 8.192 bytes.

Funciones sobrecargadas enlazadas.
Sketch uses 8.836 bytes (3%) of program storage space. Maximum is 253.952 bytes.
Global variables use 767 bytes (9%) of dynamic memory, leaving 7.425 bytes for local variables. Maximum is 8.192 bytes.

Modificación.
```

modbusSensor::modbusSensor(uint8_t id, uint16_t adr, uint8_t hold, uint8_t sizeofValue) {
  _value = new uint8_t[sizeofValue];

  _frame = new uint8_t[8];
  _frameSize = 8;
  _frame[0] = id;
  _frame[1] = READ_INPUT_REGISTERS;
  _frame[2] = adr >> 8;
  _frame[3] = adr & 0x00FF;
  _frame[4] = 0x00;
  _frame[5] = 0x02;
  uint16_t crc = calculateCRC(_frame, 6);
  _frame[6] = crc & 0x00FF;
  _frame[7] = crc >> 8;
  _status = MB_TIMEOUT;
  _hold = hold;
  MBSerial.connect(this);
}

// Constructor
modbusSensor::modbusSensor(uint8_t id, uint16_t adr, uint8_t hold) {  
  modbusSensor( id,  adr,  hold, 4);
}
```

La siguiente sobrecarga se complica  un poco pero también se podría enlazar.

peninquen, 2016-01-06:
Tienes razón, aunque depende del sketch que estemos probando. Lo había mantenido por separado para que en el caso más habitual de solo datos float el constructor sea más directo. Por contra, introduce varias versiones con lo que complica el mantenimiento.

El orden de los datos es por mantener el orden que lleva un frame modbus. ¿El comentario es por filosofía de programación o por eficiencia en el código?

cosmopaco, 2016-01-06:
Es por filosofía de programación, confunde un poco, da opción a equivocaste.

Si conoces un formato del constructor con pocas entradas con un orden X y le añades otra funcionalidad , lo logico es que pienses que se la añadiras al final.  Creo que es lo mas habitual.

peninquen, 2016-01-06:
Ok, lo corrijo, gracias.

closed 2016-01-06 by peninquen