peninquen/Modbus-Energy-Monitor-Arduino #32: Variable _framesize cual es su utilidad.

cosmopaco, 2016-01-08:
En Modbus-Energy-Monitor-Arduino/Modbus-Energy-Monitor-Arduino/ModbusSensor.cpp no se que utilidad tiene la variable _framesize. ?

peninquen, 2016-01-09:
Cosmo , estoy reordenando, pido un poco de paciencia, voy a eliminar versiones intermedias que no funcionan, eliminar variables superfluas, etc.

_frameSize la utilizo para para diferenciar entre un _frame de 8 bytes (funciones read) de un frame de funciones PRESET, que tienen 9+ sizeof(T).
La filosofia es:
 - modbusSensor almacena registros 3X y 4X,
 - los registros 3X se leen con la FC 0x04 READ_HOLDING_REGISTERS
 - los registros 4X se leen con la FC 0x03 READ_INPUT_REGISTERS
 - los registros 4X se escriben con la FC 0x10 PRESET_HOLDING_REGISTERS

El frame almacenado siempre es el de 'read'.
Solo cuando se quiere cambiar un registro 4X se construye el frame con fc 0x10, se hace la petición, y se reconstruye el frame con fc0x03.

cosmopaco, 2016-01-13:
No era mi intencion entrarte prisa ni mucho menos.

closed 2016-01-13 by cosmopaco