peninquen/Modbus-Energy-Monitor-Arduino #36: Organizar información de ``_status``

peninquen, 2016-01-20:
ModbusSensor solo almacena errores o excepciones producidas en el canal Modbus. Organizar los bits para incluir más información:

- Bit[2-0] ``slave_exception`` error comunicado por el esclavo con el código de función 0x80
- Bit[4-3] ``master_error`` error generado por el Master. Se superpone con algunos excepciones del esclavo.
- Bit[7-5] ``registered``: el valor ya ha sido leido y registrado desde la ultima vez que se actualizó. 
  
El procesador llevará dos o más procesos: 
- lectura peridica del canal ``MBSerial``.
- Publicación de los datos (MQTT, HTPP POST)
- Registro en SD.
- Actuación de relé en gestión de potencia.
- etc.

helianto, 2016-01-20:
Seria interesante generar un codigo de pulsos en caso de error, asignar un
pin a led rojo y otro a led verde y hacer una asignacion de estado en
funcion a estos.

peninquen, 2016-01-21:
¿Qué ``_status`` utiliza de referencia? es una propiedad de ``modbusSensor`` no de ``MBSerial``. Transmitir en morse pulso de 16 ``modbusSensor`` no parece adecuado.
Si conectamos varios dispositivos uno se queda offline, los otros dan ok, ¿qué código presentamos? Creo que eso se puede implementar fuera de la libreria a gusto del usuario.

helianto, 2016-01-21:
Me he explicado mal, me refiero a codigos del tipo:

Verde... todo ok
Parpadeo verde lento: envio a internet correcto
Parpadeo verde rapido: lectura de datos correcta
Rojo continuo: fallo conexion a internet
Parpadeo rojo lento: fallo conexion modbus
Parpadeo rojo rapido: datos recibidos erroneos

peninquen, 2016-01-21:
Vale, se trataría de incluir una función global que siga los estados, tanto de MBSerial como del cliente MQTT (me he decidido a profundizar en MQTT, lo veo el protocolo más adecuado para dar salida a los datos, sobre todo desde el ESP8266)

helianto, 2016-01-21:
la principal virtud es que es muy simple...