peninquen/Modbus-Energy-Monitor-Arduino #41: Ayuda con CVM - Mini

alvaro1975, 2019-01-05:
Hola! Llevo días (muchos) intentando comunicar un Arduino UNO con un CVM mini mediante un RS485 a TTL pero no me acerco ni de lejos. He consultado decenas de páginas pero con la generación de la petición no consigo avanzar.
He visto que uno de los puntos a realizar en el futuro era una versión para este analizador de red. Si has publicado algo o sabes dónde puedo encontrarlo te lo agradecería infinitamente.
Estoy intentado monitorizar una instalación fotovoltaica aislada con varios generadores de apoyo en una aplicación móvil (que ya la tengo hecha con Unity). Tiene una parte gráfica en la que se ve qué elemento es el que está generando la energía para la instalación y muestra los parámetros correspondientes.
En fin, si me pudieras ayudar, te lo agradecería infinitamente.

kluszon, 2019-01-08:
Hello Alvaro,

do you speak English?

Best regards,
Michał

kluszon, 2019-01-08:
Please use English, if you can. I recommend you to use Arduino Mega, because on Uno it isn't easy to achieve it.

ogromis, 2019-01-08:
Hello,
I speek Polish oraz English. Can you write your mail in English? This would
be helpfull for me.

Best regards,
Adam