peninquen/Modbus-Energy-Monitor-Arduino #43: Como realizar la conexion y lectura de mas de un medidor??

LeoHenrik, 2020-01-29:
He realizado la conexion de un medidor utilizando la libreria normal de sdm.h pero no logro entender como al conectar otro medidores en paralelo A y B, como deberia especificar el numero del esclavo y como extraer los parametros que requiero.

LeoHenrik, 2020-01-29:
> He realizado la conexion de un medidor utilizando la libreria normal de sdm.h pero no logro entender como al conectar otro medidores en paralelo A y B, como deberia especificar el numero del esclavo y como extraer los parametros que requiero.

closed 2020-01-29 by LeoHenrik

ogromis, 2020-01-29:
Hello,

Sorry but I dont understand anything. Can you write in English?

Regards

LeoHenrik, 2020-01-30:
I have other question, do you know how modify the ID address using arduino? or is necesary use other program for modify that?